ghaerr/elks #2383: Trouble With Assembly Programming For ELKS

TheManStudios, 2025-09-07:
System Details:
OS: Fedora 42
CPU: AMD Ryzen 9 5900X
Kernel: 6.16.5 (Same issue on 6.15.11)

After following the installation guide and familiarizing myself with ELKS, I then started on the "Writing applications for ELKS with the GNU as assembler" guide, but unfortunately I got stuck due to errors I've never seen before and couldn't figure out how to fix.

For all of this I used the build tools that are produced by the `./build.sh` script for ELKS 0.8.1, in `cross/bin`, I also tried using the same tools build by the current master branch of ELKS, but all of my results were exactly the same.

These are the steps I took:
1. I created a `test.s` file containing only the code from the hello world example
2. Using the binary in `cross/bin`, I ran this command: 
`./ia16-elf-gcc -melks-libc -mcmodel=small -c ~/projects/asm/test.s -o ~/projects/asm/test.o`
Which succeeded, as I expected
3. However, after running the next listed step: 
`./ia16-elf-gcc -melks-libc -mcmodel=small -nostdlib ~/projects/asm/test.o -o ~/projects/asm/test`
I received this error: 
`error: need runtime-specs from C library to link programs`

As I'm not particularly fluent with C programming I was confused by this, and after doing some research I was unable to find a solution to this error (the most relevant thing I tried was symlinking `/usr/include` to `~/Downloads/elks/cross`, which was what the CROSSDIR variable used by the build script was)

I then decided to try just directly using ia16-elf-as and ia16-elf-ld.gold, which went as follows:
1. I used the same `test.s` file from the previous attempt
2. Also using the binary in `cross/bin`, I instead ran this command:
`./ia16-elf-as ~/projects/asm/test.s -o ~/projects/asm/test.o`
Which succeeded, as I expected
3. I then ran this command:
`./ia16-elf-ld.gold ~/projects/asm/test.o -o ~/projects/asm/test`
Which this time, also succeeded and produced a binary 
4. However, when trying to run the produced binary using `./test` in ELKS (qemu, using the `./qemu.sh` script), it gave me this error:
./test: 1: syntax error: "(" unexpected

This was especially confusing as there are no parenthesis anywhere in my file (I tried manually re-typing it to rule out hidden characters, but it didn't remove the syntax error)

Output of `file` command on the outputted `test` binary:
`test: ELF 32-bit LSB executable, Intel i386, version 1 (SYSV), statically linked, not stripped`

The assembly code in question:
```
	.code16
	.text
	.global entry

entry:
	movw $len, %dx
	movw $msg, %cx
	movw $1, %bx
	movw $4, %ax
	int $0x80

	movw $0, %bx
	movw $1, %ax
	int $0x80
	.data

msg:
	.ascii "Hello, World!\n"
	len = . - msg
```

So it seems like it built a 32-bit binary for the i386, which makes me think I'm missing some important flags for either `ia16-elf-as` or `ia16-elf-ld.gold`, are there specific flags I'm missing or was something in my pipeline incorrect?

toncho11, 2025-09-07:
Hi, 

A few quick comments:

- we use this gcc or https://github.com/tkchia/gcc-ia16 or Open Watcom compiler
- these are C compilers, so you need to use ASM inline statements
- you can use WASM (Watcom Assembler) I think (adjusted for ELKS)
- there is also a recent compiler, assembler chain: https://github.com/ghaerr/8086-toolchain This chain allows you to build ELKS executables in C / ASM both cross compiling and natively (inside ELKS). You have:
  * as  filename.s     # to assemble .s files into object code
  * ar  filename.a *.o # to bundle object files into a library archive

toncho11, 2025-09-07:
This is an example in ASM: https://github.com/ghaerr/8086-toolchain/blob/dev86/examples/show_fonts.s
And here it shows how to compile it: https://github.com/ghaerr/8086-toolchain/blob/dev86/examples/Makefile.elks

ghaerr, 2025-09-07:
Hello @TheManStudios,

You're almost there. ELKS itself doesn't actually run ELF-32 binaries, instead requires MINIX a.out format. ia6-elf-gcc normally runs `elf2elks` on the ELF file prior to finalizing, but this isn't being done since you're running ia16-elf-as and i16-elf-ld.gold directly. Running `elf2elks test test.elks` should fix the problem. In the future, you can set your ld.gold output file to be test.elf and then use `elf2elks test.elf test` as the final step.

>  it gave me this error:
./test: 1: syntax error: "(" unexpected
This was especially confusing as there are no parenthesis anywhere in my file 

The reason for this is that the ELF file was marked executable, but wasn't a MINIX a.out executable, so the kernel rejected it, and then the shell went ahead and treated it as a BASH script file.

Thank you!

ghaerr, 2025-09-07:
> However, after running the next listed step:
./ia16-elf-gcc -melks-libc -mcmodel=small -nostdlib ~/projects/asm/test.o -o ~/projects/asm/test
I received this error:
error: need runtime-specs from C library to link programs

This error is likely because the ia16-elf-gcc compiler driver isn't setup to run by directly running from `cross/bin`. Instead, use `. env.sh` from the ELKS directory to automatically add the cross/bin directory into your path. This should then allow the ia16-elf-gcc driver to run properly, as well as find the runtime-specs. When this works, you won't have to manually run `elf2elks` or `ia16-elf-as`, etc.

When the path is set correctly using `env.sh`, you should be able to also build ELKS using `make clean; make`.

TheManStudios, 2025-09-07:
Thank for you replying, @ghaerr! This information you've provided seems helpful, however, I seem to be having issues with `elf2elks` tool.

When attempting to use the syntax you provided (`elf2elks test.elf test`) I was given the error:
`elf2elks: error: multiple file names!`

It seems this tool only takes in an input file, so I dropped the `test` at the end and ran it again, this receiving this error:
`elf2elks: error: entry point outside near text segment`

This makes me think either I'm still missing some important flags or the code provided in the guide is no longer correct for this version of ELKS. Also, maybe you were referring to a newer/older version of the tool? Do note that I'm using the version built by the ELKS 0.8.1 build script.

TheManStudios, 2025-09-07:
Thank for you replying, @toncho11! 
> * there is a recent compiler, assembler chain: https://github.com/ghaerr/8086-toolchain This chain allows you to build ELKS executables in C / ASM both cross compiling and natively (inside ELKS).

Unfortunately I had issues compiling that toolchain on my machine, though I assume I can combat this by building it within an Ubuntu docker container. If I'm unable to get my current setup working via that elf2elks tool ghaerr mentioned, I'll try this and hope it works.

toncho11, 2025-09-07:
> Thank for you replying, [@toncho11](https://github.com/toncho11)!
>
>
> Unfortunately I had issues compiling that toolchain on my machine, though I assume I can combat this by building it within an [...]

You need to follow all the information here: https://github.com/ghaerr/elks/wiki/Setting-up-the-8086-toolchain-(C86-compiler-and-tools) including some of the pull requests links.

toncho11, 2025-09-07:
I think the first step is compiling ELKS and getting used to env.sh before anything else.

ghaerr, 2025-09-07:
> It seems this tool only takes in an input file, so I dropped the test at the end and ran it again, this receiving this error:
elf2elks: error: entry point outside near text segment

Yes, my bad - `elf2elks` only takes a single input file. To make things much simpler, I advise that you use `. env.sh` to get a known working ELKS compilation capability, then use the main ia16-elf-gcc compiler driver, to develop your first assembly .S file. This can be done as follows:

C file test.c:
```
int main(int ac, char **av)
{
    return 1;
}
```
Then compile it to produce a .S output file:
```
ia16-elf-gcc -melks-libc -mtune=i8086 -Os -mcmodel=small -S main.c
```
FYI, the produced main.S will look something like:
```
    .arch i8086,jumps
    .code16
    .att_syntax prefix
#NO_APP
    .text
    .global main
    .type   main, @function
main:
    pushw   %bp
    movw    %sp,    %bp
    movw    $1, %ax
    movw    %ax,    %ax
    movw    %ax,    %ax
    popw    %bp
    pushw   %ss
    popw    %ds
    ret
    .size   main, .-main
    .ident  "GCC: (GNU) 6.3.0"
```
Then look at main.S and base your assembly off of that, meaning insert your stuff in main, then assemble it using the compiler driver:
```
ia16-elf-gcc -melks-libc -mtune=i8086 -Os -mcmodel=small main.S
```
This will produce a working ELKS (MINIX) a.out. You can then proceed to start deleting lines you don't want in main.S to produce the shortest ASM input file for your research.

There are other ways, but they are a bit complicated to explain without me having to check on various details. After playing around with this, you should be able to get to using `-nostdlib` etc.

> here is a recent compiler, assembler chain: https://github.com/ghaerr/8086-toolchain This chain allows you to build ELKS executables in C / ASM both cross compiling and natively (inside ELKS).
Unfortunately I had issues compiling that toolchain on my machine

The 8086-toolchain should compile on Linux without problems. Please open an issue over there including the error messages if you would like more help on that.

Be forewarned however, that will the way-too-many ASM input formats available for 8086 assemblers, each of these (GCC uses GAS format, 8086-toolchain uses Intel format AS86 variant, and NASM uses Intel format NASM variant, etc) assemblers will require different source code.

> receiving this error:
elf2elks: error: entry point outside near text segment

The reasons for this have to do with special linker scripts which are today oriented around C, rather than ASM programs. We can get deeper into linker scripts but I don't recommend starting with them, which is why I've reversed direction and suggesting you use ia16-elf-gcc to run all the sub drivers for you, at least in the beginning.

> I then started on the "Writing applications for ELKS with the GNU as assembler" guide, but unfortunately I got stuck

Yes, it seems that document is quite very old. Perhaps we can get a Wiki update on how to write in GCC assembler once you get this working, at least documenting the initial steps to take to get an executable from scratch.

Thank you!

ghaerr, 2025-09-08:
> My main issue with this is that it appears you use the --prefix flag when building these tools, which means I wouldn't be able to copy them out of the container and would end up having to manually read through and edit them to fix this.

I see. Well, perhaps using `./ia16-elf-gcc` as the compiler driver could still work from the cross/bin directory, as it should be able to run ia16-elf-as and the linker from within your restricted environment.

TheManStudios, 2025-09-09:
> The default ELKS linker script for small model is elks-root/elks/elks-small.ld. 
This worked! Though it did require renaming the `entry` fields in the hello world program to `_start`, as the linker script did mention that `entry` was an older name and didn't work anymore.

To anyone else that for whatever reason is using these tools directly, these are the exact steps I had to do:
(First, add the tools to your path, they should be in [elks root]/cross/bin and [elks root]/elks/tools/bin respectively)
Then run:
`ia16-elf-as -mtune i8086 -c [your code.s] -o [output name.o]`
Then:
`ia16-elf-ld.gold -T [elks root]/elks/elks-small.ld [output name.o] -o [binary name]`
And finally:
`elf2elks [binary name]`

After all of this, the binary ran perfectly fine under elks!

closed 2025-09-09 by TheManStudios