ghaerr/elks #2385: The `sys` utility writes to the wrong MBR

un-clouded, 2025-09-16:
When trying to use the `sys` command to prepare `hdb1`, the MBR for `hda` is overwritten.

The system has two HDDs.  I was trying to prepare `hdb1` (and the MBR for `hdb`) for ELKS.  I had booted ELKS from 1.44 MB floppy.  When I gave the command:

    sys -M /dev/hdb1

..I expected the MBR of `hdb` to be prepared with the boot loader, but instead `hda` was overwritten, which sadly cuzed the Devuan MBR.  This was with ELKS 0.8.0, but I've since confirmed that 0.9.0-dev does the same in Qemu.

- Version (commit or release): 0.9.0-dev
- Emulator or HW platform: i440BX / Qemu

- Does the problem always happen ?  Yes

To reproduce:
Start Qemu with two disk drives:

    qemu-system-i386 -name elks -machine isapc -cpu 486 -m 32 -nographic -drive file=/tmp/hda.img,index=0,media=disk,format=raw -drive file=/tmp/hdb.img,index=1,media=disk,format=raw -nic model=ne2k_isa -boot c

Then ask it to:

    sys -M /dev/hdb1

It says:

    Installing/updating ELKS on -M
    System on /dev/hda: Minix (CHS 63/16/63 at offset 0)
    Writing MBR on /dev/hda
    Target on /dev/hdb1: Minix (CHS 65/16/63 at offset 63)
    Bootblock written
    Copying /linux to /tmp/mnt/linux
    Copying /bootopts to /tmp/mnt/bootopts
    System copied
    Creating /dev
    Copying /bin
    Copying /etc
    Finished successfully.

(Note that the MBR on `hda` rather than `hdb` is overwritten).

ghaerr, 2025-09-16:
Hello @un-clouded,

I am very sorry to hear that our sys utility trashed your main drive MBR. I will take a look and get a fix going ASAP.

Thank you.

ghaerr, 2025-09-17:
@un-clouded, I fixed this in #2387 and tested with QEMU. Thank you.